jbrazio/ardufocus #1: SGP won't connect

kingjamez, 2017-09-22:
Hi! Thanks for putting this up for us all to use! I've got ardufocus on a Arduino Nano. From the standalone Moonlight focusing software everything works. I can also control it as expected from putty to my COM port. However, SGP won't connect to the device. When I click connect on the Moonlight DRO Driver, the configuration window pops up, but show's no controllers. When I sniff on the serial port, I see that the Ardufocus is sent :GH#:2GH# and then doesn't respond. I get :GH# being sent, but I don't understand :2GH#. When I do :GH# manually (from putty) ardufocus responds correctly. Any thoughts?

jbrazio, 2017-09-29:
You may need to put the cap to hold the reset line, without it the ASCOM driver will not work.

jbrazio, 2017-09-30:
Closing after [discussion](https://www.cloudynights.com/topic/585306-ardufocus-open-source-moonlite-compatible-focuser/?p=8125666) on Cloudy Nights forum:

> I added the cap and it's all working perfectly now. I'm using the Moonlight DRO Focuser driver and I have excellent performance on my Moonlight High Res stepper motors. I needed to Mod the steppers in the moonlight unit to be bi-polar instead of unipolar, but they are six wire stepper motors, so I only had to cut one of the six wires and then my DRV8825 controlled the motor perfectly.

closed 2017-09-30 by jbrazio