jbrazio/ardufocus #6: could you add some wiring  instructions for modded motor

djibb, 2019-04-13:
Hi, 

It's difficult for me to know which wire is where... all is A/B Black and Red... 

here for example : https://github.com/jbrazio/ardufocus/blob/master/doc/wiki/schematic-bipolar_bb.png
Which is the right wiring with colors ?
pink orange yellow, blue ?

Orange near 12V ? 5V ?

Tks a lot... I heard some n,oise but don't know if my morotr is down or ma A4998 is down... or my wiring is down...

:) tks for help a noob.

jbrazio, 2019-04-13:
Well wire colours vary per supplier, usually there is some trial and error when you don´t know the motor phases. You may get a DMM and measure the resistance between each pair of wires, if they are on the same coil then you´ll get a reading, otherwise it will show open circuit. After knowing which pair belongs to each coil is just  a question of choose one coil to be A and the other to be B.

Just keep in mind to cut all the power to the Arduino (USB cable) and the step stick BEFORE unplug any motor wires.. not doing so will fry the step stick micro.

If your motor does a humming noise and does not move then for sure is the coils in the wrong place.

Here is a table with common wiring schemas.. but yours may be different !

[image]

jbrazio, 2019-04-13:
@djibb are you using a Nema 17 or a modded 28byj-48 ?

closed 2019-05-12 by jbrazio

Portia-Lin, 2022-07-05:
@jbrazio You have not specified the contacts of the windings quite correctly. The place where the contacts of the windings are soldered to the board should have the following designations: A1 B1 X A2 B2. So:
Yellow: A1
Blue: A2
(or numbered in reverse)
Orange: B1
Pink: B2
(or numbered in reverse)