jbrazio/ardufocus #8: Ardufocus components

Ultimaboot, 2019-04-22:
Hello,

You did a great job with the Ardufocus, thanks.

I have a question about your PCB.
What means the 100nF and 10K unter the Arduino Nano board.
100nF is not a part of the components.
Are this components on the top side or bottom side of the PCB?

And what is the purpose of the NTC component?

Please could you explain it to my?

With Regards

Peter Boot

jbrazio, 2019-04-22:
Hi Peter,

When I get time I need to do a better documentation about the project.. I do understand that people have to dig a bit to find everything. Thanks for pointing that out, indeed the BOM is out of spec in relation with the PCB design.

The 10K resistor is part of the NTC logic and it's needed to act as a voltage divider, it is listed in the BOM as "1x Resistor 1/4W 10K". This resistor should be measured before being soldered into place and the value of NTC_RESISTOR_VAL set to the exact measurement, this will make the NTC to report more accurate data.

The 100nF cap is decoupling cap added on the latest version of the PCB and this is the reason it's not on the BOM.

All components on this PCB are mounted on the top side and soldered on the bottom with the exception of these two you pointed out, as they are SMD they are also soldered on the top surface.
I had past requests to make this a DIP only board meaning all SMD should be removed and replaced with trough hole ones this is somewhere on the pipeline but I've been busy getting the ASCOM driver and dual-motor support working.

The current PCB version is also lacking new functionality such as `ENABLE_DTR_RESET` which will disable/enable the RESET hold capacitor by a serial command thus allowing USB firmware upgrade without removing the Nano from the PCB.

Ultimaboot, 2019-04-23:
Thanks for the quick replay.
The preforated Board is than also not up to date?
Is there now a new version of the PCB available?
If not than i have to remove the Nano from the PCB, upgrade the software and than put it back to the PCB, correct? Could i use a PCB arduino nano socket strip to overcome this problem for this moment?

jbrazio, 2019-04-23:
Both are out of date in relation with the DTR reset.

You can use a socket and remove the nano for programming or with a bit of practice you can press the reset button at a specific time when uploading from the IDE/PlatformIO and it will proceed.

Ultimaboot, 2019-04-23:
Last question for now:)
what voltage is needed for the 100nF component, 50V?

jbrazio, 2019-04-23:
The voltage on that rail will be 5V, so anything above 10V will work.

BTW, if you are building the prefboard version it's easy to enable the DTR function, just connect the GND leg of the 10uF cap to pin A5 of the Nano and then set the `DTR_RESET_PINOUT` to 15, you also need to enable `ENABLE_DTR_RESET`.

Ultimaboot, 2019-04-24:
Thanks for the reply and tips.

Ultimaboot, 2019-04-24:
What is the PRGM on the right upper corner?

jbrazio, 2019-04-25:
That’s the old method to enable the automatic reset for firmware upload. If you cut the trace then the reset will be enabled, put a solder blob and it’s locked again.

This auto reset is a big thing when using the Moonlite ASCOM driver, it will fail to connect to the focuser. With my new drivers it is irrelevant.

Ultimaboot, 2019-04-25:
If i want to use your PCB i only have to put all the components in the BOM and the one we spoke of in the mail before/above and connect with a piece of wiring the GND leg of the 10uF cap to pin A5 of the Nano. 
Do i have to cut the trace (PRGM) yes or no? With this question answered i think this will work, correct?. 
Setting the DTR_RESET_PINOUT to 15 is a software adjustment, correct?
Enable ENABLE_DTR_RESET, is this cutting the trace (PRGM) or is this also a software adjustment?

jbrazio, 2019-04-26:
If you want to use the PCB as is you have all the components spaces there so it will be OK.

If you want to use the DTR feature with that PCB you will need to cut the trace on the PGRM jumper and solder a wire between the left side and pin A1 (Example) then activate on the firmware config.h the feature and the Pin you selected.

But.. this feature is yet experimental and it will be there for Moonlite driver compatibility only.

My advice is: solder some female headers on the PCB and pop up the nano if you want to upgrade the firmware. To make it easy to remove the nano, populate only the nano board with the required male header pins, leaving all the other (non used) pins bank. If you are unsure which ones should be populated I can give you a list. This will make a lot easier to pop up the nano board from the PCB.

Ultimaboot, 2019-04-26:
In the mail of 25-4 your mention connecting to pin A5 of the Nano board and now it is pin A1 of the Nano Board. Is it pin A5 or A1?
a list would also be helpful, thanks.

jbrazio, 2019-04-26:
It’s just an example you may use A1 or A2 or any other one available.

A0 is pin 14 (used by NTC)
A1 is pin 15
(...)
A5 is pin 19

closed 2019-05-12 by jbrazio