jbrazio/ardufocus #9: Pro Micro 32U4 support possible?

kyuzumaki, 2019-05-07:
Would it be possible to support the Arduino Pro micro?

I tried adding it to the list of boards but the registers seem to be different. Is there an easy fix for this or is it a limitation of the chip?

```
Board: "SparkFun Pro Micro, ATmega32U4 (3.3V, 8 MHz)"
serial.h:50: error: 'UCSR0A' was not declared in this scope
         UCSR0A &= ~bit(U2X0); // baud doubler off
```

jbrazio, 2019-05-07:
The ATmega32U4 was not on the supported list of microcontrollers; I don't have any on hand but I'll update the HAL and you'll do the validation.

jbrazio, 2019-05-07:
Are you positive your board is the 3.3V at 8MHz and not the 5V at 16MHz ?

If it is the 3.3V version there are some things to consider:
- Timers should be OK as they are calculated based on `F_CPU`
- NTC will work as it is VCC independent.
- 3.3V logic may have issues with the step stick drivers which are 5V logic.

jbrazio, 2019-05-08:
@kyuzumaki Do you mind to test ?
Download the tarball from [here](https://github.com/jbrazio/ardufocus/archive/hal/atmega32u4.zip), uncompress, configure and upload.

Pay attention to the weird numbering on the pins..
`A0 is 18`, `A1 is 19`, `A2 is 20` and `A3 is 21`.

kyuzumaki, 2019-05-08:
Tried with CoolTerm but the com port just dosen't show up at all.

I tried to add in some Serial.println("DEBUG INFO") lines into the code to see where it jams but the command is not recognised. I'm guessing that the default serial driver is different somehow from what the Nano uses. The [sparkfun pack](https://cdn.sparkfun.com/assets/learn_tutorials/1/2/1/SparkFunBoard_Addons_Jan_16.zip) has a few Hardware_Serial.h files in there but haven't figured out whats different.

jbrazio, 2019-05-08:
You didn't answer to my question, do you have the cap between GND and RST ?
Maybe it needs special initialization, I'll check how the Sparkfun variant is handling the serial port.

`Serial()` is not available because I'm not using the Arduino framework. Try the following, get and LED with proper resistor on `PIN A3` then edit `ardufocus.cpp` and at the end of the `DEBUG` section after the `#endif` add `DDRF = bit(PF4)` then you can toggle the LED using `PORTF ^= bit(PF4);`

Validate first with a blank sketch that the LED is working:
```C++
void setup() {
  DDRF = bit(PF4); // Set A3 as OUTPUT
}

void loop() {
  PORTF ^= bit(PF4); // LED ON
  delay(1000);
  PORTF ^= bit(PF4); // LED OFF
  delay(4000);
}
```

jbrazio, 2019-05-09:
I've been digging into the Arduino core source code and this will be a bit harder than I initially thought, at a first glance all the USB "core" subsystem needs to be implemented so the Serial can work..

So the device has two serials, one is the USART1 and the other is the USB based one. The changeset I've build targets the internal USART1, so the serial which is available at pins `TX0` and RX1` which will need a USB-to-TTL converter in order to be used. :disappointed:

@kyuzumaki If you have a FTDI USB-Serial 3.3V dongle can you confirm that the USART1 is working ?

kyuzumaki, 2019-05-09:
Sounds like this is a more complex problem than I had thought initially! I’ve just ordered an Arduino Nano since it has the separate serial chip and works already.

I do have an FTDI serial board but it’s not with me at the moment. I can test the second serial port it in a few weeks when I’m back at my workshop but I’m starting to think it may not be worth adding support for the chip unless there is more demand for it. 

In answer to your question about the  capacitor I don’t have any 10uF caps to try the largest I have at hand is 100nF not sure if that’s big enough to work?

jbrazio, 2019-05-09:
If you don't mind validating the USART1 when possible I would appreciate it and would keep the HAL supporting that bit, at least what was done is kept. Full USB Serial support is another story for now.

100nF should work also.

kyuzumaki, 2019-05-10:
Have tested using USART1 connection and it works. Was able to connect it to kstars and control the 28byj stepper motor.

jbrazio, 2019-05-11:
Thanks for the time taken to validate the PR #10 at least not everything is lost.

closed 2019-05-11 by jbrazio