jbrazio/ardufocus #11: Modified step motor 28BYJ-48 with PCB not working

Ultimaboot, 2019-05-12:
Hey,
I finally succeeded in assembling the PCB. I soldered the components on the PCB, and put the code in the Arduino Nano.
Put 12V power supply on PCB and the PWR LED lights up green. The usb connected to the laptop. Arduino Nano is found by the laptop on COM 6. When I connect to the software, the DX and RX Led start blinking at the same time. When I try to control the motor, the flashing speed of the DX and RX LED changes briefly and then returns to the original speed. The step motor does not move at all. The step motor is a modified 28BYJ-48 and is connected according to issues 2 of this site. The step motor also gets hot. Can you give a clue as to what I can do?

jbrazio, 2019-05-12:
Hi @Ultimaboot !

Try to first check if the firmware is OK, just open Teraterm or miniterm session (9600bps, 8-bit, no parity, 1 stop bit) and you should be greeted with a welcome message, if not try to press the `reset` button on the Nano while being connected to the serial port.

If the motor gets hot usually there is and issue with the motor windings. There has been a lot of confusion due to the messing naming schema (1A, 2A, 1B, 2B). I have not corrected [this post](https://github.com/jbrazio/ardufocus/issues/6#issuecomment-482882163) and published a [new guide about modding the motor](https://ardufocus.com/docs/28byj-48-bipolar-hw-mod/#modding-the-motor). Please read it and double check your connections.

Ultimaboot, 2019-05-12:
Connecting with tera term gives me the following message:

```
 Ardufocus 0.2b-master ready.
                            Visit https://github.com/jbrazio/ardufocus for updates.
```

jbrazio, 2019-05-12:
Then the firmware is OK.

jbrazio, 2019-05-12:
That's a good sign, the firmware should be OK.
Did you had time to check the motor windings ?

Ultimaboot, 2019-05-12:
The motor windings are correct.
Orange connected to pin 1A A4988;
Pink connected to pin 1B A4988;
Yellow connected to pin 2A A4988;
Blue connected to pin 2B A4988.
Red not used and middle trace step motor by scrapping the PCB varnish and copper.

jbrazio, 2019-05-12:
Could you post your `#define MOTOR1_PINOUT` from `config.h` ?

Ultimaboot, 2019-05-12:
#ifdef MOTOR1_USE_A4988_DRIVER
  //                                       MS1, MS2, MS3, SLEEP,  STEP, DIR
  #define MOTOR1_PINOUT  12,   11,    10,      8,         7,      6
#endif

jbrazio, 2019-05-12:
Are you using the PCB version or the prefboard one ?

Ultimaboot, 2019-05-12:
PCB version

jbrazio, 2019-05-12:
To use `OUT1` on the silkscreen do `#define MOTOR1_PINOUT 16, 6, 5, 4, 3, 2`to use `OUT2` do `#define MOTOR1_PINOUT 12, 11, 10, 9, 8, 7`.

Ultimaboot, 2019-05-12:
Its not working.

jbrazio, 2019-05-12:
Do you care to take a overview picture from the top side of the board with everything connected ?

Ultimaboot, 2019-05-12:
[image]

jbrazio, 2019-05-12:
Everything looks OK. I assume that 2nd wire is orange one and not the red one, on my motor orange's color tone is yellowish rather than reddish as yours.

Did you adjust the pot on the step stick ?
Do you see the flat lip on the pot ? Try rotating it 180º CW.

Ultimaboot, 2019-05-12:
Yes I did try to adjust the pot on the step stick CW and CCW but that didn't work.

jbrazio, 2019-05-12:
Do you hear or feel any vibration from the motor when it was supposed to be running ?
Which software are you using to test the motor ?

Ultimaboot, 2019-05-12:
I do not hear or feel any vibration from the motor.
I use MoonliteSingleFocuser_v1.4 to test the motor.

jbrazio, 2019-05-12:
[image]

- Test continuity between the top pins in purple and the Arduino pins.
- Test continuity between the bottom coil pins and the motor wires.

- Measure resistance between "1B 1A" and "2A 2B", which values do you have ?

- Measure voltage between RST and GND you should have +5V
- Measure voltage between ENA and GND you should have 0V
- Measure voltage between bottom two first pins, you should have +5V
- Measure voltage between bottom two last pins, you should have +12V

- Is the NTC connected between 1 and 2 or 2 and 3 and not between 1 and 3 ?

jbrazio, 2019-05-12:
Could you please also post your entire `config.h` ?

Ultimaboot, 2019-05-13:
Test continuity between the top pins in purple and the Arduino pins. This is correct.

Test continuity between the bottom coil pins and the motor wires. This is correct.

Test continuity between ENA and GND you should have it. This is correct.

Measure resistance between "1B 1A" and "2A 2B", which values do you have ?
1B (Pink) 1A (Orange) = 203 ohm and 2A (Yellow) 2B (Blue) = 201 ohm. This a 12V step motor nr 181228766

Measure voltage between RST and GND you should have +5V. This is correct

Measure voltage between bottom two first pins, you should have +5V. This is correct

Measure voltage between bottom two last pins, you should have +12V. This is correct

Is the NTC connected between 1 and 2 or 2 and 3 and not between 1 and 3 ? 1 and 2 (left and middle on picture)

Config.h (ZIP file)
[config.zip](https://github.com/jbrazio/ardufocus/files/3174117/config.zip)

jbrazio, 2019-05-13:
The config file looks OK also.
All tests check out.

Can you test continuity between M1B and all others ?
Can you test continuity between M1A and all others ?
Can you test continuity between M2A and all others ?
Can you test continuity between M2B and all others ?
Can you test continuity between the two parts of the cut trace on the motor ?

If all this checks validated then I must start thinking about a broken step stick.
Do you have special tools such as a logic analyser or an oscilloscope ?

Ultimaboot, 2019-05-14:
Can you test continuity between M1B and all others ? This is correct.
Can you test continuity between M1A and all others ? This is correct.
Can you test continuity between M2A and all others ? This is correct.
Can you test continuity between M2B and all others ? This is Correct.
Can you test continuity between the two parts of the cut trace on the motor ? This correct.

Can you test continuity between any ground pin on the arduino and the first bottom left pin on the step stick ? (This is the ground pin of the +5V on the step stick). This is correct.
Test again between GND on arduino and the 7th pin on the step stick (This is the +12V ground). Both should have continuity. This correct.

Do you have special tools such as a logic analyser or an oscilloscope ? I have no special tools.

If all this checks validated then I must start thinking about a broken step stick. Well i was smart to buy some spare parts. When i Exchange the step stick nothing happened but i did hear some noise from the step motor, so i turned the pot and now it is working. So try it again with old step stick and nothing happened. The step stick was indeed broken. Is there a ideal setting for the pot?

jbrazio, 2019-05-14:
Great news !

These step sticks are very susceptible to burn out if we accidentally, or not, disconnect one of the four motor coils with it having +12V power.. it will instantly burn out internally.

There is a trick to correctly set the pot to your motor power requirements, so you have a 200R per coil motor at 12V this gives us 0.06A, let’s round it to 0.1A.

Use the following formula:

Vref = 0.1 * 8 * 0.1 <=> Vref = 0.08V

Take a multimeter and measure the positive on the center of the pot against any GND point on the board. You should set this to the Vref value above.

**Attention this is assuming that the little SMD resistor near the pot is a 0.1R one.**

Could you please check the number written on it ?

Ultimaboot, 2019-05-15:
The Resistor on the A4988 is R100 what is 0.1ohm.
Vref = 0.1 * 8 * 0.1 <=> Vref = 0.08V.
The setting of the pot is now 0.08V.
Everything is still running.
Next step 3D printing
I will close this issue.

closed 2019-05-15 by Ultimaboot