jbrazio/ardufocus #12: compile error - error: no matching function for call to 'uln2003::pinout_t::pinout_t(<brace-enclosed initializer list>)'

jrhilton, 2019-05-16:
Hi, i've downloaded the files and simply opened ardufocus.ino and configured for ULN2003 per the readme. When trying to compile it comes up with the following error (using Arduino IDE 1.6.9).

What am I doing wrong?

ERROR:

In file included from sketch\motordrv.h:31:0,

                 from sketch\ardufocus.h:30,

                 from sketch\ardufocus.cpp:20:

sketch\uln2003.h: In constructor 'uln2003::uln2003(const uln2003::pinout_t&)':

uln2003.h:56: error: no matching function for call to 'uln2003::pinout_t::pinout_t(<brace-enclosed initializer list>)'

     inline uln2003(pinout_t const& pinout) : m_pinout({ pinout }) { ; }

                                                                 ^

sketch\uln2003.h:56:65: note: candidates are:

sketch\uln2003.h:32:12: note: uln2003::pinout_t::pinout_t()

     struct pinout_t

            ^

sketch\uln2003.h:32:12: note:   candidate expects 0 arguments, 1 provided

sketch\uln2003.h:32:12: note: constexpr uln2003::pinout_t::pinout_t(const uln2003::pinout_t&)

sketch\uln2003.h:32:12: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const uln2003::pinout_t&'

sketch\uln2003.h:32:12: note: constexpr uln2003::pinout_t::pinout_t(uln2003::pinout_t&&)

sketch\uln2003.h:32:12: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'uln2003::pinout_t&&'

exit status 1
no matching function for call to 'uln2003::pinout_t::pinout_t(<brace-enclosed initializer list>)'

jbrazio, 2019-05-17:
Hi @jrhilton,

I've just tried to compile with IDE 1.8.5 and was successful using `#define MOTOR1_USE_ULN2003_DRIVER`:
```
Sketch uses 10608 bytes (36%) of program storage space. Maximum is 28672 bytes.
Global variables use 512 bytes (20%) of dynamic memory, leaving 2048 bytes for local variables. Maximum is 2560 bytes.
```

Could you attach your `config.h` file ?

jrhilton, 2019-05-17:
Hi there, I think the error is at my end, I tried compiling on another laptop with the latest version of the IDE and it worked without an issue. I suspect something is wrong with the older install. This can therefor be closed as solved. Many thanks for the prompt reply, sorry I seem to have wasted your time.

closed 2019-05-17 by jrhilton