jbrazio/ardufocus #14: Moonlite Compatibility

bbillp, 2019-07-07:
How Is the Adufocus controller connected to the Stand alone Moonlite Hi-Rez Stepper motor DB9 connector ?  I don’t see circuits for the DB9 connections.   Thank you .....

jbrazio, 2019-07-08:
Hi !
Have a look at the INDI documentation for the Moonlite Protocol [here](http://indilib.org/media/kunena/attachments/1/HighResSteppermotor107.pdf).

[image]

To connect to Ardufocus you need to use the ULN2003 driver because the Moonlite motor is a unipolar one, have a look [here](https://ardufocus.com/manual/build-instructions/#motor-connections).

bbillp, 2019-07-08:
Thank you for this information, So the Moonlite Motor assembly DB9 goes direct to the Stepper motor so I understand there is NO electronics interface in the Moonlite motor box, only the stepper.  Now the project makes sense to me.     

I want to use the Moonlite Motor because it has some very nice mechanical features.

Do I use the Arduino IDE to load your code ?

Will you include a LCD display for Stepper Position like the PNP_Focus project ?

Are you in Portugal, I love the country and drove my car from Faro, Salame in the South thru Lisboa, Porto, Nazare, Obidos and north, very very nice ....

Thank you  ..... Bill P. In Huntington Beach, California USA

jbrazio, 2019-07-08:
Hi BIll,

Before going into expensive motors give a try to a cheaper solution.. you may be surprised. ;-)
I've been using a prototype focusing mechanism with my TS MonorailR96, even when using single stepping on the driver it can archive sub micron movement (0.195 micron) for each step or in other words the resolution is around 5 step/micron.

[image]

If there is demand for the LCD we can work it out.

Yes I'm indeed from this little paradise named Portugal. :-)
Did you had the opportunity to stay a night in Alentejo ? There's were we have the best starry nights.

bbillp, 2019-07-09:
Yes I agree on a simple solution like your drawing but the end of the lead screw must not interfere with the diagonal or the camera.  The focuser need not have a long travel as focus will be preset by eye then use the focuser to ‘fine tune’ the image. My drawing shows an idea to use a push rod and a lever to turn the focus knob a few degrees for fine tuning.  The push rod can be motivated by one of 3:  Stepper, linear actuator or RC airplane servo which is very cheap.  

Is Ardufocus code loaded with the Arduino IDE?  I have not studied the code today.

I made a small linear actuator using a 3mm threaded rod and modified RC servo motor and searched the internet for a NEMA-8 or smaller stepper last night, some micro steppers are only $2.00, amazing.  Old PC printers and scanners are very good sources for these parts and there are many of these things in the trash so I have a treasure box of small motors and will use the DRV8825 in my final version IF necessary.

I believe the Focuser can control any one of the motors except the servo, maybe .... as you know the RC servos operate by frequency and PWM modulation, simple 5 volt one wire signal + power supply 5V.  

It should not be too difficult to load PNP_Focus LCD code into Ardufocus, I will try it after loading Ardufocus but I thought your code included LCD, its been a long day and no time to study.   I feel a LCD with counter readout is very necessary when using a manual controller.  A photo of PNP_Focus is below.

We often visit Madrid Spain then drive somewhere.  During the 90s we drove to Cadiz then east to Faro, Lagos and when I got tired of driving I found a little road to the beach, Salame, wonderful ! We had a room right on the beach for two nights.  Over the next 6 nights we worked our way up the coastal route (Portugal is very similar to Southern California) staying in Nazare, Lisboa, Porto, Obidos then north to Santiago de Compostella Spain.  We saw the beautiful castles in Cintra and Pena (?) and ate LOTS of seafood.  My son returned numerous times to Portugal near the Spanish border for short holidays.  We have wonderful memories of our trip and before I get too old I want to return.

Siesta time, I look forward to hear from you ..... Bill

PNP_Focus Arduino

jbrazio, 2019-07-09:
Your message didn't had any attachments.

Indeed, I can get away with such a rod in that position because I have a thin-long imaging train: Reducer - OAG - FW - CCD

Yes it should compile with the Arduino IDE but the code itself is not "arduino" based, so libs that are made for Arduino will not work with Ardufocus without some knowledge.

bbillp, 2019-07-09:
Hello Joao, I don’t know why the attachments disappeared but here they are again.    Bill

Astrotech AT80edt with focus lever drawing with push rod by servo or linear actuator.

Home made Mini Linear Actuator

PNP_Focus readout.

jbrazio, 2019-07-09:
You’re using the email to reply to the thread, I don’t think they will work like that, tru using the github web page instead.

closed 2019-07-31 by stale[bot]