jbrazio/ardufocus #18: Sometimes focuser goes to 0 absolute, instead of step out or step in command.

jedas, 2019-12-01:
Quite often, like 5% of the time, focuser goes to 0 absolute position instead of making small step. It looks like that it does that more often if step button is pushed whileUsing Ekos/Indilib on virtual machine. Arduino mini. Can't be sure that it's not a driver's issue.

jyecker, 2019-12-17:
I have almost the same problem. It does't go to 0 but will go to a low absolute value. Something like 26, 48,or 228.  It doesn't matter whether it's a small in or out move that's commanded.  It behaves the same with Moonlite ASCOM or Ardufocus ASCOM or the Ardufocus Handpad app. It will not stay connected to the Moonlite Windows app. It usually disconnects after  GT# but not right away. I'm using a Mini Pro so I don't have an NTC hooked up (no AREF pin) but I tried it on a Nano with an NTC temp sensor and I get the same results. It just seems random. Checking off the Trace box the commands from the the computer seem OK.

jedas, 2019-12-18:
I had a few imaging sessions with firmware which limits it to 1000 steps max. Sometimes focusing fails, but it never jumps so far into back focus (low values), thus it's way easier to work and it's more stable. Not using NTC either.

jbrazio, 2020-01-08:
First of all sorry for the late reply.
@jedas @jyecker Are you using ASCOM with my driver or the Moonlite one ?

>  It will not stay connected to the Moonlite Windows app. It usually disconnects after GT# but not right away.

This is usually due the arduino auto reset feature, did you install the capacitor between GND and the RST pin ?

jedas, 2020-01-08:
I'm using Indilib (EKOS) on VM Ubuntu on Windows 10. FTDI virtual serial port for USB-TTL connection and arduino mini.

jbrazio, 2020-01-08:
> Quite often, like 5% of the time, focuser goes to 0 absolute position instead of making small step. Using Ekos/Indilib on virtual machine. Arduino mini. Can't be sure that it's not a driver's issue.
> Kind of 'solved' by checking target distance, if it's >1000, don't move in function stepper::set_target_position.

But this is not a solution.. target distances can be a lot bigger than 1k step.
Do you have an option on EKOS to log the serial communication with the focuser ?

I see that @jyecker has a different setup from yours, he is using ASCOM and you INDILIB.. so we may be looking at a true bug here. May I ask you @jyecker to do the same thing, on the ASCOM settings activate the serial debug and send me a log file where the issue occured ?

jedas, 2020-01-08:
I've enabled debug log check, but I didn't see any communication log there. I'll look for it during next imaging session. I was suspecting, that some command byte is not being recognized correctly, therefore ending up as zero. From time to time, I still see EKOS displaying 0 coordinate, as reported by focuser. But this is being filtered with my hack, focuser doesn't move to that location, so it doesn't corrupt focusing procedure. I understand that it's not good solution to limit to 1k step, but it kind of worked on my setup. I'm looking forward for proper fix. Your firmware is great.

jyecker, 2020-01-08:
Here's my setup. I'm using the ULN2003 motor driver, I have commented out the buttons and the LED's. So about as stripped down as I can get. I'm using a Mini Pro with the FTDI cable. I have not had the cap hooked up and I don't have a temp sensor hooked up either. I did have a Nano with a sensor and a cap hooked up but it responded the same. The Mini Pro will connect with your app, your ASCOM driver and the MoonLite ASCOM driver without any timeout problems. They all have the run away problem. The MoonLite app is the only one with the disconnect issue. You'll see in the log file that after a command to goto 2025 (SN07E9) it ends up at 233. 
Here is the log file from ascom. 
[ASCOM.MoonLite.1729.497710.txt](https://github.com/jbrazio/ardufocus/files/4037662/ASCOM.MoonLite.1729.497710.txt)

jbrazio, 2020-01-09:
@jyecker Indeed the log clearly shows the erroneous behaviour.

Is this always reproducible ?
Did you do any changes on the codebase other than on the config file by enabling/disabling `#defines` ?

jbrazio, 2020-01-09:
@jyecker do you care to open the ASCOM Diagnostics application and enable on the Trace menu:
- Serial Trace Enabled
- Include Serial Trace Debug Information

Let's see if this gives us more information on the log file.
The file(s) should then be located at `(..)\Documents\ASCOM\Logs XXXX-XX-XX`

jyecker, 2020-01-09:
I think this is what you want, I ran it again this time it took a little longer ( time to fail is random ) but it usually fails within a few minutes or sometimes in a few clicks. I just keep making in or out relative movements.  It's pretty consistent except for the number of commands until it does it.   15:18:12.944 may be the time of interest ?  There are 2 more serial debug files if you need them. I don't think I have any code changed except the config.h.  I used the uln2003 mtr drive, changed the pins to 3,4,5,6 and commented out the UI buttons and LEDS. I do have two inputs floating. The UI_KAP_ADC_CHANNEL 1 and the A0 temp pin. The Mini Pro does not bring the A_REF of the ATMEGA out to a pin.  I did try it with a Nano and two 10k resistors to give it a constant temp but it didn't make a difference. I'm not that interested  having the temp comp. right now, although it would be nice in the future.  I thought that having that pin float might be an issue but the temp seems to be fairly consistent in the log files.  
Also the Mini Pro does have a DTR pin on the serial connection, it has RTS instead. I don't know if that makes a difference on the way it is reset by the software. Anyway the cap doesn't seem to make a difference with the ASCOM driver. It doesn't help the MoonLite app either. 
I really appreciate you taking the time to look at my issue. 
[ASCOM.MoonLite.1514.293870.txt](https://github.com/jbrazio/ardufocus/files/4042568/ASCOM.MoonLite.1514.293870.txt)
[ASCOM.Serial.1514.294040.txt](https://github.com/jbrazio/ardufocus/files/4042572/ASCOM.Serial.1514.294040.txt)

jyecker, 2020-01-10:
Don't spend too much time yet. I took it home and on my home computer I couldn't get it to do it. Let me test on some other computers first before I get you too involved. I've tested a lot of different focuser projects on my work computer so maybe sometime is corrupt. I'll update you on my results.

jbrazio, 2020-01-10:
But we have two different people with the same strange behavior.. I suspect something is there and happens on very special circumstances.

Btw, did you use the Arduino IDE to compile the code ? Can I have which version did you guys used ?

PS: I didn’t had the opportunity to check your latest logs, will do it soonish.

jbrazio, 2020-01-14:
I've checked the logs and I don't see anything to write home about.. :-/
How do you have the FTDI breakout board connected to the Nano, could it be noise on the serial line ?

I will add a special `#define` to disable temperature code block completely, this will remove some ISR/ADC workload and return a constant bogus temperature value.

jyecker, 2020-01-14:
I have two configurations. One is a Pro Mini with an FTDI Cable. Like this one. FTDI Cable 5v The other is a Nano with a CH340 USB to Serial converter on the board. ( Chinese Arduino clone) Not an FTDI but built into the board.It might be helpful to leave out the temp stuff since it 's not going to be easy to do with the Pro Mini without some rework.The behavior is the same with both setups. The USB2-USB3 thing seems not to be the case. I don't know why it seemed to work that one night.Have you had any experience with the cheap Arduino's and do you think that could be the issue?Thanks again for your efforts. John

| 
| 
| 
| $17.95 |  |

 |

 |
| 
|  | 
FTDI Cable 5V

The FTDI cable is a USB to Serial (TTL level) converter which allows for a simple way to connect TTL interface d...
 |

 |

 |

John

jbrazio, 2020-02-07:
@jyecker I use a lot of cheap 1 dollar nanos from Aliexpress and my experience with them is rather positive, in fact both of my focusers are using those cheap arduinos and they work just fine.

I suspect this could be a compiler error. Each Arduino IDE version comes with it's own buildroot, gcc and glibc version.. which means whe can have different binary outputs from the same source code.

Could you please tell me your Arduino IDE version ?
Do you know how to standalone flash a binary to the Arduino using avrdude ? I could compile it using my environment and then check if it behaves the same on your hardware.

jyecker, 2020-02-07:
I'm using Arduino IDE 1.8.10.
I haven't used AVRDUDE before. I looked at a ladyada tutorial and although I think I could do it, It may require some babysitting on your part. You may not wish to do that which I would fully understand. I'm not a coder and rely on the generosity of guys like you for my code which is greatly appreciated. 
Just to clarify, this is not reproducible on your end?

closed 2020-02-29 by stale[bot]