jbrazio/ardufocus #19: A way to specify step size in UI_KAP

ccaramel, 2020-04-26:
Hi,
congratulations for this amazing work!! even I got some issues using Moonlite protocol.

The questions is, I'm currently using UI_KAP function, which is working, but It move step by step at a speed determined by Arduino polling which is very very slow.
Is there a way to interact with this or modify defaults in your code?
Regards.

jbrazio, 2020-05-09:
Hi !
Do you want to specify a "fixed" speed at build time or have a way to change it on-the-fly when using ?

ccaramel, 2020-05-10:
Hi, finally succeeded in setting a fixed value in ui_keybd.cpp then succeded in using a pot to change focus speed. Maybe it could be useful to join a schematic with pot connections. 
The only remaining problem is with acceleration during manual focusing, it would be great to be able to set an acceleration profile dedicated to ui_keybd.

Using MoonLite protocol (both with indi and indigo) it regularly shows an absolute position @0 and a temperature @0 for a few seconds then come back to real value. It has the same behavior using various nanov3 or uno. What do you opine about it?
Regards

closed 2020-06-01 by stale[bot]