jbrazio/ardufocus #20: Speed with the  user interface (switches).

Antiath, 2020-10-24:
I would like to change the speed in manual mode. Righ now, the acceleration is very slow and the top speed achieved is also quite low. It is fine though when I'm controlling the device with the ascom driver ( the firmware is set to 400 steps/sec maximum in the config file). Wich part of the code do I have to change to resolve this ?  
Also, is this manual mode using half steps ou full steps by default ?

Antiath, 2020-10-24:
Well never mind. I figured out what that #define UI_KAP_ADC_CHANNEL line was for :p Now I just need to add a pot to ro the remote.
As a side note, it woul be nice to document that speed control feature. The config.h file doesn't give much info about that. It doesn't even state the possibility to control the speed of the user interface.

closed 2020-10-24 by Antiath

dwilbourn618, 2022-01-17:
Antiath, It's nice that you worked out what the  #define UI_KAP_ADC_CHANNEL line was for, but not all of us are as smart as you! Can you explain a little about what this does and how I can use it to control the manual speed?

jbrazio, 2022-01-20:
@Antiath thank you for the comment. Indeed the documentation is lacking.. the key feature was a bit slapped into the source code.

@dwilbourn618 you need to connect a potentiometer to and ADC channel and define that channel on the config using `define UI_KAP_ADC_CHANNEL`, then the position where the pot is at will set the speed that focuser will move when a key is pressed.

max2206, 2022-02-25:
> @Antiath thank you for the comment. Indeed the documentation is lacking.. the key feature was a bit slapped into the source code.
>
> @dwilbourn618 you need to connect a potentiometer to and ADC [...]

OK to which pins need one to connect a pot