jbrazio/ardufocus #21: Ardufocus components

ApoEvans, 2020-11-16:
Hello João,

I want to build the arduino focuser using your code and schematics as a reference. 
Can I succeed without using the resistance and thermistor and if not could you please elaborate on their function on the project ? 
I do not have a thermistor at hand so, can I substitute thermistor with a resistance ?

Best Regards

Apo

jbrazio, 2020-11-19:
Hi Apo,

Their purpose is to report back temperature to the ASCOM driver which in turn is used by the imaging program to (re)focus your imaging train every X degrees delta in temperature.

The original Moonlite has builtin on the focuser firmware an automatic temperature compensation algorithm which does not rely on the imaging software that will automatic move a x amount of steps based on temperature delta.. but this doesn't work that good and I decided not to implement it.

I would say that the system will even work without the components but the best would be to populate the thermistor location with a 10K resistor, it should always report back 25C.

ApoEvans, 2020-11-19:
Hi João, I will try it with resistor you suggested. Thank you, Apo Sent from Mail for Windows 10 From: João BrázioSent: Thursday, November 19, 2020 1:40 PMTo: jbrazio/ardufocusCc: ApoEvans; AuthorSubject: Re: [jbrazio/ardufocus] Ardufocus components (#21) Hi Apo,Their purpose is to report back temperature to the ASCOM driver which in turn is used by the imaging program to (re)focus your imaging train every X degrees delta in temperature.The original Moonlite has builtin on the focuser firmware an automatic temperature compensation algorithm which does not rely on the imaging software that will automatic move a x amount of steps based on temperature delta.. but this doesn't work that good and I decided not to implement it.I would say that the system will even work without the components but the best would be to populate the thermistor location with a 10K resistor, it should always report back 25C.

closed 2020-12-12 by stale[bot]