jbrazio/ardufocus #24: Ardufocus is not recognized by the Moolite ASCOM driver

nikivan1, 2021-03-07:
Compiled and uploaded the firmware to Arduino Mini. I am able to communicate with it from the Moonlite Single Focuser 
 v1.4 app, but the Moonlite DRO Focuser Driver (downloaded from focuser.com) does not show any devices in the Moonlite Focus Setup menu. I am running the latest version 6.5 of the ASCOM Platform.

nikivan1, 2021-03-07:
Found about the reset capacitor. With 100nF between Reset and D3 and the appropriate config settings it works.

closed 2021-03-07 by nikivan1

bakosmi, 2022-07-12:
please what config settings? i have the same issue and struggle to resolve it. many thanks!

jbrazio, 2022-07-15:
You need to add a capacitor, otherwise the arduino will reset at every serial connection and the moonlite driver will not recognize the focuser as connected.

bakosmi, 2022-07-15:
i have added salvaged led diode between reset and 5v and it works fine - many thanks! :-)