jbrazio/ardufocus #32: focuser jumps to zero position quite randomly while searching for focus INDI, EKOS

astroberlin, 2023-01-18:
Hi João, thank you for the great project.

I have two focusers running 2a on an UNO and controlled by INDI Moonlite DRO driver in EKOS, Kstars.

During focusing, the focuser will randomly jump back to the zero position, as if it is misreading a command.  Any settings/timings I can optimize to prevent this?

Thank you.

Mr-Groch, 2023-07-23:
I'm experiencing the same issue also on ASCOM driver (moonlite and native)

ChrisN4gel, 2024-06-23:
Hi João, thanks for Ardufocus from me as well... I love it.
I also have the issue with the position being 0 sporadically. I am using it on two rigs with China-Unos (CH340 Driver from NanjingQinhengMicroelectronics) and the stepper motors are both 28BYJ-48 / ULN2003

But anyway, great project which brought motorized focussing to my telescopes.  :-)
Thanks, Christian